bambulab/BambuStudio #6331: Enter key on 10 key pad not recognized

wyowarren, 2025-03-30:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

P1S

### How to reproduce

Enter key on 10 key keypad not recognized in this version. For instance, use the Move command to change the coordinates of a model on a plate, change a number using the 10 key and press Enter. The number is changed but the change is not accepted until you press Enter on the main keyboard.

### Actual results

See description in How to reproduce

### Expected results

See description in How to reproduce

### Project file & Debug log uploads

See description in How to reproduce

### Checklist of files to include

- [ ] Log file
- [ ] Project file

guanyun-gudujian, 2025-03-31:
will be fixed next version

guanyun-gudujian, 2025-04-16:
@wyowarren 
fixed
pls try the latest version: https://github.com/bambulab/BambuStudio/releases/download/v02.00.02.57/Bambu_Studio_win_public-v02.00.02.57-20250415200919.exe