bambulab/BambuStudio #6333: OBJ import

Tommo681, 2025-03-30:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11 pro

### Additional system information

AMD Ryzen 5 5600 6-Core Processor                 3.50 GHz
32.0 GB DDR4 Ram
Nvidia Geforce Rtx 3060 12GB

### Printer

Bambu lab A1/AMS combo

### How to reproduce

Created an obj in Image-to-3D from MakerWorld.
It wont import into Bambu studio.

### Actual results

Bambu Studio Crashes.

### Expected results

OBJ should import.

### Project file & Debug log uploads

[crash_Sun_Mar_30_19_37_40_0_done.log](https://github.com/user-attachments/files/19527082/crash_Sun_Mar_30_19_37_40_0_done.log)

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-03-31:
Hello, can you provide this obj？
Can you tell me which website is Image-to-3D

Tommo681, 2025-03-31:
Website. https://makerworld.com/en/makerlab/imageTo3d?from=makerlab
OBJ wont send over email, file too big

________________________________
From: Haidiye00
Sent: Monday, March 31, 2025 08:54
To: bambulab/BambuStudio
Cc: Tommo681; Author
Subject: Re: [bambulab/BambuStudio] OBJ import (Issue #6333)

[Haidiye00]Haidiye00 left a comment (bambulab/BambuStudio#6333)<https://github.com/bambulab/BambuStudio/issues/6333#issuecomment-2765419049>

Hello, can you provide this obj？
Can you tell me which website is Image-to-3D

Tommo681, 2025-03-31:
Its ok, ive sorted it out. the OBJ from website was too big. I scaled it down if Fusion 360 and it works now.

closed 2025-03-31 by Tommo681