bambulab/BambuStudio #6334: AMS filaments group empty when connecting to the printer for the first time in session

Skydev0h, 2025-03-30:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 (10.0.26100.3194)

### Printer

X1C, P1S

### How to reproduce

1. Start the application
2. Go to "Device" tab
3. Select printer with AMS
4. Go to "Prepare" tab
5. Click the filament selection dropdown

### Actual results

AMS filament section is present, but is empty

The filaments appear only if another printer is selected, and the one in question is selected again
I do not know how this workaround would be possible if there is only one printer

### Expected results

AMS filament section is present and is populated

### Project file & Debug log uploads

[log_Sun_Mar_30_22_46_35.zip](https://github.com/user-attachments/files/19527330/log_Sun_Mar_30_22_46_35.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

Skydev0h, 2025-03-30:
Problem demonstration

https://github.com/user-attachments/assets/0efcc340-ef35-44cd-85cc-bbc06114f14c

Skydev0h, 2025-03-30:
By the way just noticed another bug in the video... PLA Wood is recognized as "Generic PLA" o_O

Skydev0h, 2025-03-30:
Update: It seems that the bug only happens if "Multi device management" is activated.

walterwongbbl, 2025-04-09:
Hi. @Skydev0h  
The multi-device management may cause subscription information to fail. If multi-device management is turned off, everything works fine.