bambulab/BambuStudio #6339: Ams change of filament not saving

simonth3dactions, 2025-03-30:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1C, Bambu Lab P1S

### How to reproduce

How to Reproduce

1. Open either the Bambu Handy app or Bambu Studio.
2. Attempt to change the color or filament type, for the printer it self on the AMS.
3. Observe that the selection briefly updates for a few seconds.
4. The settings revert back automatically shortly after.
5. When using Bambu Studio on PC, the change simply doesn’t save.
6. When using the Bambu Handy app, a message saying "fail" appears, and the changes are not saved.

### Actual results

The change of filament selection is not saved

### Expected results

It should save like it always does.

I have 8 printers, and this has never happened before. I have of course restarted the printer, ams and also checked everything has the latest firmware.

I could still start the printers, but the color mapping just had to be double checked, since the colors in the ams are not correctly represented.

### Project file & Debug log uploads

It doesn't really have anything to do with the project is self, since the problem is with the change of colors it self.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

simonth3dactions, 2025-04-01:
RESOLVED

Here is what I did to fix it - (Everything on the same version of Bambu Studio (2.0.1.50) on a Windows 11 PC)

1. Turned of the printer(s)
2. Removed any AMS connection to the printer
3. Turned the printer back on
4. Changed the color on the external spool in Bambu Studio and checked if that got saved. Which it did
5. Connected only one AMS to the printer, and let it finish the whole start up process (rolling in filament etc.)
6. Changed the color on just one slot in Bambu Studio to see if it saved. Which it did
7. Then did that with the other slots
8. I then connected the second AMS, and also changed the colors and filament settings in Bambu Studio. It all saved

NOTE!
I’m not sure if this is related to the issue, but I had another printer working perfectly fine and was able to change the color in Bambu Studio on my PC. However, after I changed the color in the Bambu Handy app, that printer started experiencing the same problem.

My theory is that the Bambu Handy app may have sent a "faulty request," triggering an action in the printer that caused the error.

I can’t say for certain if that’s the cause, but to resolve the issue, I stopped using the Bambu Handy app entirely and relied only on Bambu Studio on my Windows 11 PC.

I’m hoping the Bambu Handy app will work again soon, as I use it in my workflow to change the filament color and settings via my phone.

griches, 2025-04-10:
The steps here worked for me, but to simplify I just had to restart the P1S and then only use Bambu Studio to change filament type in the AMS. If Bambu Handy is used then it stops the AMS having a different filament assigned until it restarts.

simonth3dactions, 2025-04-14:
Yeah, that is also the case for me I can see.

That is quite weird, that the app suddenly doesn't work like that anymore...

walterwongbbl, 2025-04-21:
Hi. @simonth3dactions 
Could you please provide some images or the Studio logs?

simonth3dactions, 2025-04-21:
I have actually just tried out the app again, and this time changing color worked. I do not know if it is just temporarily fixed, but it worked for me now:)

griches, 2025-04-21:
I believe this has been fixed in the latest Bambu Handy and is listed under "what's new". 

[image]

simonth3dactions, 2025-04-22:
Okay awesome!