bambulab/BambuStudio #6340: Plug-in fails to install Mac

Kimsfx, 2025-03-31:
### Bambu Studio Version

Bambu_Studio_mac-v02.00.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 11.7.10

### Additional system information

[image]

Version Bambu_Studio_mac-v02.00.00.95 worked perfect installing the plug-in but after the update it no longer will install the plugin to connecct

### Printer

Bambu X1e

### How to reproduce

Try to open and install new software

### Actual results

Plugin fails to  install and there are none of the software installed on my system listed in the help web page

### Expected results

I expect it to install as it does with 1.10.02.95

### Project file & Debug log uploads

nothing to add

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Kimsfx, 2025-03-31:
> Hello, we will fix it. Thank you for your feedback.
> At present, you can restart Bambu Studio again.

Thanks..  It took a few times to get it working but im good now and
discovered a new bug in v2.

If you have 2 or more AMS attached and do a Sync, it adds spools that do
not exist,  In my case I have 2 AMS units fully loaded and when I sync to
update, it adds 4 more spools (ex spool 9 - 10 -11 & 12) but I only have 2
AMS units.

Thoughts?
Kim Bailey
(818) 370-1577

Haidiye00, 2025-04-01:
Hello, can you provide the reproduction process of GIF? Or screenshots are also acceptable

Kimsfx, 2025-04-01:
I tried to reproduce the error are it did not do it this time.  when i launched the app, it did an update so i am not sure if that fixed it.  i wll change a spool to see if it does it again

closed 2025-04-16 by Haidiye00