bambulab/BambuStudio #6341: Failed loading configuration file - Bambu X1C not shown

Felos9001, 2025-03-31:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Built from source by yourself or others

### OS version

Arch Linux (Garuda Dragonized)

### Additional system information

_Arch Linux | Linux 6.13.8-zen1-1-zen

CPU   - AMD Ryzen 5 7600X (12) @ 5.45 GHz
GPU  - NVIDIA GeForce RTX 4060 Ti [Discrete]
GPU - AMD Raphael [Integrated]
RAM - 16 GiB (40%)

### Printer

Bambu X1C

### How to reproduce

1. Install Bambu studio from the AUR using yay package
2. Open Bambu Studio 
3. Bambu studio displays error "  Failed loading configuration file /home/felos/.config/BambuStudio/system/BBL/filament/Bambu PLA
Matte @base.json
Suggest cleaning the directory /home/felos/.config/BambuStudio/system firstly"

This I think causes a preset error in the Bambu X1C as the printer does not display in the printers

### Actual results

[image]

[image]

### Expected results

I should be able to use the printer presets or at least connect to the printer

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19528133/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2025-03-31:
yes, please just follow the suggestion

close bambu studio and remove /home/felos/.config/BambuStudio/system
then reboot bambu studio

Felos9001, 2025-03-31:
I tried that however the same error occurs and it would recreate the system repository when I would open the application. 

It somehow fixed itself by deleting and reinstalling it from the AUR (Archlinux user repository), so Im just gonna continue on. 
Thank you!