bambulab/BambuStudio #6342: Rotation of Cut plane is wrong

tboy-123, 2025-03-31:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 1`0

### Additional system information

not related to issue

### Printer

Bambulab X1 C

### How to reproduce

1 import object in Bambu studio
2. try to cut it, rotate the cut plane in different axes.  Let say first in Y direction, then X then Z
3. rotation goes wrong
4. it was OK in  v.1... of Bambu studio

### Actual results

rotation of cut plane in different axis goes wrong.

### Expected results

Rotation of cut plane in different axes to be like in version 1 of Bablu studio.

### Project file & Debug log uploads

[after  Y and Z rotation X is tilting the plane-not rotating in proper rotation.zip](https://github.com/user-attachments/files/19530067/after.Y.and.Z.rotation.X.is.tilting.the.plane-not.rotating.in.proper.rotation.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

guanyun-gudujian, 2025-03-31:
will be fixed next version

guanyun-gudujian, 2025-04-16:
@tboy-123 
fixed
pls try the latest version: https://github.com/bambulab/BambuStudio/releases/download/v02.00.02.57/Bambu_Studio_win_public-v02.00.02.57-20250415200919.exe

tboy-123, 2025-04-18:
Fixed ! Thank you for fast response!

closed 2025-04-18 by tboy-123