bambulab/BambuStudio #6344: Import 2 obj files as single object with multiple files

Lasse193, 2025-03-31:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 Pro

### Printer

Ba,bu X1C

### How to reproduce

Create a file with 2 solids and export them as OBJ, on import as single object with multiple files it shifts the first object to -2,5, -2,5 (files are working in the versions before the last big update)

### Actual results

I can move the one solid to 0,0

### Expected results

No shifting of parts as in the old versions

### Project file & Debug log uploads

[Machine Tag 5602-114-03_PETG White_100.zip](https://github.com/user-attachments/files/19530262/Machine.Tag.5602-114-03_PETG.White_100.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-03-31:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

Lasse193, 2025-04-01:
[image]

This is the best i can do.. Just tested the files on another PC, with the same install, and here it works as supposed to... I just uninstalled the complete bambu lab, reinstalled, tried to login, but got only a white screen (seen it as another bug), so installed the old version, logged in, and the installed the new version. Same problem.
Tried other files, and here's the shift different (in X and Z direction)
Thanks
Lasse

Haidiye00, 2025-04-01:
[image]
Hello, I have this effect after importing the latest software. What is the problem with this?

Lasse193, 2025-04-01:
Nothing there, but look at my GIF, there the text is not in-place, as i wrote, its working on one PC, but not on another

Lasse193, 2025-04-01:
[image]

Lasse193, 2025-04-01:
With these files et was 2,5 on both x and Y direction, 
With another set it was 52 in x and 4,83 in Z
but its the same every time i import the same set (
I downgraded 1.10.2 on the same PC and now the same files import correctly

Haidiye00, 2025-04-16:
Hello, please use the latest version, thank you.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.02.57

Lasse193, 2025-04-16:
Just tested it, still the same.

Haidiye00, 2025-04-16:
Hello, actually I haven't quite understood what you mean yet. Can you explain in more detail. Can be combined with some screenshots~

Lasse193, 2025-04-16:
On other PC's the text import in the right position, as well on your's

On mine, the second imported obj shifts position (import them at the same time as "single object with multiple parts")

If i use the same file on an older version of BS on the same PC the import corrrectly, so something changed in the new version

Haidiye00, 2025-04-16:
[image]
So when you mean offset, which one has occurred~

Lasse193, 2025-04-16:
See the text in my picture... its not imported in the right position

Lasse193, 2025-04-16:
It's not imported as on your picture

Haidiye00, 2025-04-16:
[image]
Hello, I haven't popped up your window before. Could you please send me the mtl file missing from your obj file.

Lasse193, 2025-04-16:
[Machine Tag 6062-254-01_PETG Black_100.zip](https://github.com/user-attachments/files/19771732/Machine.Tag.6062-254-01_PETG.Black_100.zip)

I included the 3mf file as well so you can see how they import on my PC

Haidiye00, 2025-04-16:
Hello, I roughly understand what you mean. For now, you can delete two MTLs and import them, and the ones with overlapping positions should still work. With MTL, there is indeed a slight deviation between the two positions

Lasse193, 2025-04-16:
Super, just tested it... and even better, deleting the MTL's meant no color matching (very nice for me)

Haidiye00, 2025-04-17:
Hello, thank you for your patient explanation. I have identified the issue and resolved it. Thank you again for your feedback~

[image]

closed 2025-06-03 by Haidiye00