bambulab/BambuStudio #6345: Unable to send print - "Uknown nozzle"

zivbot, 2025-03-31:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

Dell Precision 3660
Connected to printer via LAN

### Printer

Bambu Lab P1S (no ams)

### How to reproduce

Prep any print and try to "print plate", whether I choose Generic PLA / TPU

### Actual results

An error message appears: "Printing high temperature material with Unknown nozzle may cause nozzle damage.."
"Send" is disabled.
Attempted to change nozzle diameter, choose another material, and reselect the printer type.
[image]

### Expected results

It should allow printing.

### Project file & Debug log uploads

[Cube.zip](https://github.com/user-attachments/files/19531016/Cube.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-03-31:
Thank you for your feedback. Please set the nozzle type and diameter in the "Printer Parts" section under the "Device" page. Try initiating the print again.
[image]

apolunar, 2025-03-31:
I have the exact same issue, and also cannot change the options in the dialog. Please fix ASAP, I spent 30 minutes trying to print via LAN before giving up and grabbing the micro SD card. Again, this problem was not present before the update.

zivbot, 2025-03-31:
@apolunar while they work it out, you might want to reinstall  V1. It seems to be working fine.

BoBoYu-1, 2025-04-01:
OK, sorry. Could you please try again on the machine? After setting it on the machine, it will be synchronized to the software. Please refer to the following diagram for the settings.
[image]

zivbot, 2025-04-01:
Hi @BoBoYu-1 , it seems my printer does not have an "Accessories" option anywhere in the menu.
I have firmware v 01.04 and can't update it.
Any way to disable the nozzle validation from the software side?

XinZhangBBL, 2025-04-10:
@zivbot Sorry for the issue. Could you please reboot your printer, and retry sending print

tjbland-saumag, 2025-05-14:
+1, same boat as OP.  I'm on firmware 1.04.02 and don't want to upgrade.  But there is no accessories menu option and in Bambu Studio the options are greyed out like the [screenshot on above post](https://github.com/bambulab/BambuStudio/issues/6345#issuecomment-2765992583).  I suspect this'll be closed as "resolved in later release" and ignored.  Bummer for those of us who want more control over the device.

LordIrishDave, 2025-06-29:
I have the same issue.  Printed fine last night.  Would not print today.  Zero changes made to the printer.

apolunar, 2025-06-30:
If you use Flatpak and need to reinstall V1 to get remote printing working again `flatpak update --commit=a6ffe9b20dd2b52e800f2f0c340a0d1fe2dfc2cfd9907a1a63816e8b44757261 com.bambulab.BambuStudio` worked for me.

zivbot, 2025-07-27:
> We apologize for the inconvenience caused. The new version may have resolved your issue. Please give it a try: https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60

Hi @XinZhangBBL , thank you for sharing the update. I'm still unable to send a print. We need a way to manually set the nozzle type from the software side, without the firmware update. 
Or, at least be able to bypass the "exceeds the hardness of the nozzle..." warning.

[image]

XinZhangBBL, 2025-07-28:
@zivbot Thank you for your feedback.
（1） The wiki doesn't help?

[image]

（2） To get more information, could you reproduce the problem to offer log. you can refer to https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

zivbot, 2025-08-03:
@XinZhangBBL  I'm trying the nozzle setup process - it crashes each time i click "Extruder" in the printer settings window. Attaching log file.

[bambu-studio.exe.36264.zip](https://github.com/user-attachments/files/21564291/bambu-studio.exe.36264.zip)

XinZhangBambu, 2025-09-24:
@zivbot Did the software crash when you were using the Preparation page? Does it also crash when using the latest version?  https://github.com/bambulab/BambuStudio/releases/tag/v02.02.02.56

> @XinZhangBBL I'm trying the nozzle setup process - it crashes each time i click "Extruder" in the printer settings window. Attaching log file.
> 
> [bambu-studio.exe.36264.zip](https://github.com/user-attachments/files/21564291/bambu-studio.exe.36264.zip)