bambulab/BambuStudio #6349: User filament profiles do not transfer over on H2D

amadino13, 2025-03-31:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 Version 10.0.26100 Build 26100

### Additional system information

Processor	AMD Ryzen Threadripper PRO 7975WX 32-Cores, 4001 Mhz, 32 Core(s), 64 Logical Processor(s)
Installed Physical Memory (RAM)	64.0 GB

### Printer

Bambu P1S / Bambu H2D

### How to reproduce

When I select my P1S or X1C that I configured, I can see my calibrated filament profiles. As soon as I switch to H2D, all my user presets are gone. I believe they should transfer over.

### Actual results

H2D user filament presets are gone

### Expected results

H2D filament presets transfer over from the other printers

### Project file & Debug log uploads

-

[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Rulevoid, 2025-03-31:
This is a bummer. Just noticed this too. Thanks for submitting the bug for us all!

BoBoYu-1, 2025-04-01:
Thank you for your feedback. The reason why X/P can both view the traffic ratio in the user configuration is that the X/P models are preset to be interoperable. However, the preset settings of X/P models are not compatible with those of H2D models. Therefore, it is normal to not be able to see the traffic ratio, and this is not a bug.

Rulevoid, 2025-04-01:
> Thank you for your feedback. The reason why X/P can both view the traffic ratio in the user configuration is that the X/P models are preset to be interoperable. However, the preset settings of X/P [...]

What does 'traffic ratio' have to do with filament profiles not working across printers? Poor translation or something?

Master-Mace, 2025-04-19:
Don't understand this logic either.