bambulab/BambuStudio #6350: AMS not mapping correctly

rickfurlong, 2025-03-31:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 home

### Additional system information

Device name	
Processor	11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz   2.40 GHz
Installed RAM	8.00 GB (7.65 GB usable)
Device ID	DFC8FCD1-51CC-42D0-B325-FD01A0AD184E
Product ID	00342-20713-72069-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

### Printer

(2) P1S with three (3) AMS units attached to each printer

### How to reproduce

Used newest firmware update since release of H2D. Since that update, the AMS units do not map correctly when synchronizing the filaments in the  prepare tab in Bambu Studio. I have to manually click on each filament to produce the drop down menu and select each filament individually. The filaments are listed correctly and in the correct slot in the drop down menu and in the AMS mapping on the devices tab. When I synchronize in the prepare tab I will get anywhere from 12 to 17 filaments listed and I only have three AMS units connected to the printer. I have two P1S printers with three AMS connected to each printer. 

### Actual results

The filaments map properly in the drop down menu in the prepare tab and in the devices tab but do not automatically synchronize properly when you   auto populate with the synchronize filaments option.

[image]
[image]
[image]

### Expected results

After downloading the project file, choosing the printer and synchronizing the filaments, the filaments should be the correct filament loaded in the AMS, show the correct number of filament options and map the filaments into the correct AMS slots. 

### Project file & Debug log uploads

[Checklist_with_slider_6967384.zip](https://github.com/user-attachments/files/19536005/Checklist_with_slider_6967384.zip)

[debug_Sun_Mar_30_20_58_22_16708.log.zip](https://github.com/user-attachments/files/19536152/debug_Sun_Mar_30_20_58_22_16708.log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

JeffGoldner, 2025-03-31:
Yes, very annoying and I sometimes forget to override when I Print Plate. I have a P1S and one AMS, synchronize gives me 7 filaments including the external spool.

I'm sure this has been reported already but it's a major issue.

Haidiye00, 2025-04-01:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)
Also, your picture is not very clear. Could you please provide it again for us to analyze and reproduce? Thank you

NeomimGH, 2025-04-01:
I've just gone back to the old version of Bambu Studio and wow did it feel good. AMS mapping seemed like a dream, 2 clicks and it was synced and then my non Bambu filament added. So easy!

I was struggling to work out what was going on with the new v2 mapping and then when BBS started to bomb out when I tried to use the overwrite I gave up. So glad I did.

I'm sure it will get fixed but for now I'm sticking with v1.

Haidiye00, 2025-04-02:
Hello, you can use the v2.0.1.50 zip package as an experience. If there are any areas where the development is not suitable, please help us point them out. If it is reasonable, we will continue to improve. Thank you very much

NeomimGH, 2025-04-02:
@Haidiye00 Hi, this was the version I was using 2.0.1.50 but I had so many issues I really found it very frustrating. 
I will reinstall and feed back to try to help the community but I had to go back as I simply couldn't get a print going easily enough and it became too frustrating. 
I will post my issues.

Haidiye00, 2025-04-02:
Hello, I'm very sorry. I suggest using the free installation version to experience the new version first. Then the issues you discovered can be posted here, and we will continue to follow up

atulkhatri, 2025-04-12:
Is there any resolution to this issue? Not able to map any filament type/color in AMS and it is causing flushing issues when printing with incorrect colors.

rickfurlong, 2025-04-12:
When synching click the overwrite option and they map correctly. This didn’t really “fix” the issue but it works. Rick Furlong

Haidiye00, 2025-04-16:
Hello, please use the latest version, thank you.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.02.57

atulkhatri, 2025-04-16:
Thank you. I created a support ticket and I was told to restart the printer, which did the trick. I think it wasn't related to Bambu Studio and probably the printer/AMS.

Haidiye00, 2025-04-16:
So, has the latest version already solved your problem?

Haidiye00, 2025-04-29:
https://github.com/bambulab/BambuStudio/releases/tag/V02.00.03.54
Please try this