bambulab/BambuStudio #6352: Bed flickers with Mac version

cntbill, 2025-03-31:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Mac Pro  osv 10.15.7

### Additional system information

 Model Name:	Mac Pro
  Model Identifier:	MacPro5,1
  Processor Name:	6-Core Intel Xeon
  Processor Speed:	3.46 GHz
  Number of Processors:	1
  Total Number of Cores:	6
  L2 Cache (per Core):	256 KB
  L3 Cache:	12 MB
  Hyper-Threading Technology:	Enabled
  Memory:	32 GB
  Boot ROM Version:	144.0.0.0.0
  SMC Version (system):	1.39f11
  SMC Version (processor tray):	1.39f11

### Printer

P1S

### How to reproduce

just view build plate

### Actual results

Build plate flickers

### Expected results

Should not flicker

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-04-01:
Thank you for your feedback. Could you provide a screen recording to describe your problem?

Haidiye00, 2025-04-02:
hello,you can use QuickTime Player to record gif

Haidiye00, 2025-07-18:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~

closed 2025-07-22 by Haidiye00

cntbill, 2025-07-22:
Not really… I just have a crappy mac according to you guys

 

Closed #6352 as completed.

Haidiye00, 2025-07-22:
Hello, it seems that your Mac version is relatively low. Should you upgrade to a newer version?

cntbill, 2025-07-22:
Well according to specs Studio is supposed to work but… my computer is old from 2010, highest OS version for the year and if I had at least $2500. that I didn’t know what to do with,  I’d get a new one… but for now I have to deal with it as I don’t do windows…. Thanks for your help

GiammAstro, 2026-08-29:
Hi, i actually have the issue on a 2025 MB Air, so it is not related to old HW it seems.

cntbill, 2026-08-29:
Well I am still using the same old Mac Pro but now using Studio 2.8.2.61 and haven't had any bad flickering that I noticed