bambulab/BambuStudio #6354: Improper tree supports generation/output with 2.1.0.50...prints perfect in 1.10.2.76

andyschwartz, 2025-03-31:
### Bambu Studio Version

2.1.0.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 14.6.1

### Additional system information

CPU: 3.6 GHz 10-Core Intel Core i9
GPU: AMD Radeon Pro 5700 XT 16 GB
Memory: 128 GB 2667 MHz DDR4

### Printer

X1C, P1S - Happens on both

### How to reproduce

Open .3mf file used since January with no issue
Slice/Print File
Wait for weak stringy tree supports.
print fails every time

### Actual results

[image]

[image]

### Expected results

This is after going back to 1.10.2.76 install. This is the supports I expect to receive:

[image]

[image]

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19537284/log.zip)

[Project Export - Bambu Lab.3mf.zip](https://github.com/user-attachments/files/19537288/Project.Export.-.Bambu.Lab.3mf.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

andyschwartz, 2025-03-31:
Due to copyright/Patent issues I cannot post the project file publicly. Let me know how to get around that.

[image]

[image]

NothFie, 2025-04-01:
@andyschwartz Thanks for your report. It's hard to infer the cause only with the pictures and parameters you provided. I tried on similar models with the params and they are printed well. Have you modified any other parameters(not only in "Support")? Also could you please tell me which filaments you are using？This will help us analyze the issue.