bambulab/BambuStudio #6360: Bambu Studio won't run on Windows 11

Juanrar3d, 2025-03-31:
### Bambu Studio Version

2.00.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11  virtualized with parallels 20.

### Printer

Bambulab X1C

### How to reproduce

 1. click open bambu studio
 2. not open.

### Actual results

Until the version of bambu Studio 1.10.1.50 worked without problems from the following updates has not worked again.

### Expected results

BambuStudio not running

### Project file & Debug log uploads

### Checklist of files to include

- [x] Log file
- [ ] Project file

lanewei120, 2025-04-01:
dup from https://github.com/bambulab/BambuStudio/issues/6014

Juanrar3d, 2025-04-07:
> Fecha: lunes, 7 de abril de 2025, 9:54
>
> Hello
>
> please download this file
> win_02.00.01.03.zip<https://github.com/user-attachments/files/19627997/win_02.00.01.03.zip>
> and unzip it into your plugin directory of bambu studio
> (C:\Users\user.name\AppData\Roaming\BambuStudio\plugins)
>
> image.png (view on web)<https://github.com/user-attachments/assets/2518c019-3e8a-4595-b976-7fcca79a02a0>
>
> then restart bambu studio, it should be ok
>
> please have a try with this way currently
> we will fix it in the next version
>
>
>
> Hello
>
> please download this file
> win_02.00.01.03.zip<https://github.com/user-attachments/files/19627997/win_02.00.01.03.zip>
> and unzip it into your plugin directory of bambu studio
> (C:\Users\user.name\AppData\Roaming\BambuStudio\plugins)
>
> image.png (view on web)<https://github.com/user-attachments/assets/2518c019-3e8a-4595-b976-7fcca79a02a0>
>
> then restart bambu studio, it should be ok
>
> please have a try with this way currently
> we will fix it in the next version

Hello

I followed your instructions and the latest version of Bamboo Studio is working on my PC.

Thank   you  so  much for your solution and dedication.

Best regards.