bambulab/BambuStudio #6361: Unable to reduce the size of the purge tower since Bambu Studio version 2

Gougouille, 2025-03-31:
### Bambu Studio Version

2.0.50 (french)

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 24H2

### Additional system information

AMD Ryzen 9 7940HS w/ Radeon 780M Graphics        4.00 GHz
DDR5 32,0 Go

### Printer

BAMBULAB A1 Combo

### How to reproduce

0n The Left Menu
1-Goto "autre"
2-Section "Tour de nettoyage"
4-"Premier Volume" disappeared, making it impossible to reduce the size of the tower, which is annoying for large prints 

### Actual results

none of the new settings allow to decrease the first volume which would allow to reduce the size of the tower. the direct consequence is that the tower and the print overlap. (See image 1),

[image]

 I found a trick to move the tower after slicing (you can move the tower on the Y axis only after, see image 2 and 3), but it is not viable because I can no longer publish my creations on makerworld, the saved .3mf does not keep the position of the tower after the 2nd slicing

[image]

[image]

### Expected results

In Bambu Studio V1, you could adjust the width and first volume, which had the effect of reducing the tower so that it did not interfere with the print (See image 4)

[image]

### Project file & Debug log uploads

[Horloge Stade Rochelais simplifié.zip](https://github.com/user-attachments/files/19539943/Horloge.Stade.Rochelais.simplifie.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

zhimin-zeng-bambulab, 2025-04-01:
[image]
@Gougouille You can uncheck “rib wall” to restore the previous tower shape.

Gougouille, 2025-04-01:
I tryed all options. unchecked all, try all parameters, without success. I am still unable to reduce the volume off the tower as before. uncheck “rib wall” just remove the outside wall, but “first volume” still not present, the tower is always the same size and cannot be reduce. I don’t want to remove it, just reduce it and move it on the extreme top left of the build plate as before to be able to print and post my creations for other usersJérôme GILLET

  image.png (view on web)

image.png (view on web)

Cahmarfil, 2025-04-02:
Yeah. I'm having exactly same issue and in my case the prime tower came giant!

JeffGoldner, 2025-04-02:
Just tried on Mac version 2.0.1.50 and unchecking Rib Wall allowed changing the tower width.