bambulab/BambuStudio #6364: Bambu Studio crashing after update to 2.0.1.50

BobalouieS, 2025-03-31:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: Intel(R) Xeon(R) W-2223 CPU @ 3.60GHz
RAM: 256GB
Display Adapter: NVIDIA Quadro RTX 4000

### Printer

Bambu Lab X1C

### How to reproduce

1.  Open a complex 3mf file
2.  Slice the file
3. Change to filament color scheme in the slicing result box

### Actual results

No error message, bambu studio just shuts down and goes away.

### Expected results

bambu studio would change to the filament color scheme and not shut down.

### Project file & Debug log uploads

[Crash.zip](https://github.com/user-attachments/files/19542003/Crash.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

TamiRose, 2025-03-31:
Having the same issue on MAC, it just keeps crashing. Tried Orca, and it crashes at the same time which for me is when adding a SVG.

ProjectInitiative, 2025-04-01:
```bash
** (bambu-studio:23933): CRITICAL **: 21:49:50.925: Cannot register URI scheme wxfs more than once

** (bambu-studio:23933): CRITICAL **: 21:49:50.925: Cannot register URI scheme memory more than once
free(): invalid pointer
```

Testing 3 versions back on linux crashes on all version because of the network plugin auto downloading and causing a crash loop. Deleting the plugin allows the app to open, but as soon as it is installed and you try to login (or it uses your cached credentials) it crashes and points to the network.so file.

QingZhangBambu, 2025-04-01:
and please upload 3mf file

BobalouieS, 2025-04-01:
> and please upload 3mf file

I cannot upload the 3mf file. It is proprietary.

BobalouieS, 2025-04-01:
> Thanks，Could you please provide the DMP file? The steps to obtain the logs are as follows：https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

There was no dmp file. Only the two files that I uploaded. It failed a bunch of times and just shutdown and did not log a dmp file

BobalouieS, 2025-04-01:
I should also mention, similar to TamiRose above, I reverted back to 2.0.0.95 hoping that would fix the issue and it still persists.  That was with a full uninstall, wipe all user data, and restart the machine.  I even tried going back to 1.10.02.76 to no avail.  If you slice then try to change to any other color scheme it locks up briefly and then shuts down, sometimes it will show the color change before locking up and shutting down but usually not.

GuyH77, 2025-04-01:
I've a similar issue. Trying to slice a Cinderwings baby rose dragon for the H2D and it crashes each time. Attached are my dmp logs. I can't upload the stl for obvious reasons. 

[log.zip](https://github.com/user-attachments/files/19557465/log.zip)

Witawat, 2025-04-02:
My program has similar problem, it can't be used at all.

zhimin-zeng-bambulab, 2025-04-08:
If you can provide the 3mf or stl file that caused the slicing crash, it will help solve the problem.

JohnSmithCYM, 2025-04-10:
Can any one send your stl or 3mf file to malkey.chen@bambulab.com ? We promise the data from you will only be tested in internal environment and no data leak will happen.