bambulab/BambuStudio #6366: MacOS crash after just opening the application

ChaserGames, 2025-04-01:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.4

### Additional system information

Apple M3 Pro SOC 
18 GB Ram

### Printer

A1

### How to reproduce

Open a .3MF file in finder 
At some point will give an error message that gyroid is not a supported 100% infill
Click on any option (even closing the pop-up)

### Actual results

Software crashes and never resolves, requires a full re-install of Bambu Studio

[image]

### Expected results

Probably just go to the home-screen or open the selected file

### Project file & Debug log uploads

[Archive.zip](https://github.com/user-attachments/files/19544585/Archive.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-04-01:
Hello, could you provide a reproduced operation video or GIF? Thank you

ChaserGames, 2025-04-01:
https://github.com/user-attachments/assets/b6d319ba-630d-4923-9aa7-d1c6c2c5eb5c

ChaserGames, 2025-04-01:
https://github.com/user-attachments/assets/f989d76d-7047-4c39-9fc4-06366831af54

Haidiye00, 2025-04-03:
Hello, can you please re upload the 3mf file that you clicked to open? Thank you

closed 2025-04-16 by Haidiye00