bambulab/BambuStudio #6367: No possible Register on Bambu Studio

Asirus4499, 2025-04-01:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro, version 24H2

### Additional system information

Intel(R) Core(TM) i7-10510U CPU @ 1.80GHz   2.30 GHz
8,00 GB (7,83 GB iskoristivo)
64-bitni operacijski sustav, procesor x64

### Printer

Bambu Lab A1

### How to reproduce

Go to Login/Register
Open empty Window withouth anything ( white square )

### Actual results

No possibility to register on software product

### Expected results

Register and Login

### Project file & Debug log uploads

No Project

### Checklist of files to include

- [ ] Log file
- [ ] Project file

kobidavid-tech, 2025-04-01:
i wanna join to that bug im getting this problem also

Asirus4499, 2025-04-01:
What You mean another network?

Asirus4499, 2025-04-01:
Maybe is problem with Your software for win 10. another with win 11 have problem

Asirus4499, 2025-04-03:
I try with another network. Nothing

realzorro, 2025-04-16:
Please choose language under Bambu Studio settings, this issue is resolved.