bambulab/BambuStudio #6374: Filename format not being respected

davidjdixon, 2025-04-01:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.4

### Printer

H2D

### How to reproduce

1. Enable 'Develop mode' in settings,
2. Modify the 'Filename format' under 'Others' in the global process settings
3. Add any model/primative to project
4. Use 'Export plate sliced file'

### Actual results

The filename in save dialogue is the default filename format

### Expected results

The filename in the save dialogue should match the format provided in 'Filename format'

### Project file & Debug log uploads

N/A - easy to replicate

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-04-02:
Hello, after modifying the 'Filename format' under the 'Global Parameters' -> 'Others', the changes will only take effect for the file names in the 'File' -> 'Export' -> 'Export Gcode' section; for other export methods, the file names will use the default ones.
[image]

davidjdixon, 2025-04-02:
@BoBoYu-1, thanks for your response. This is inconsistent with how I've been doing this with Orca Slicer, but I appreciate that it is a forked project. How do go about changing the file name for exported plates in that case, as exporting GCcode for bambu printers is rather pointless as its the 3MF sliced file we would be exporting more often (especially for print farms where it's often more practical to copy files to USB drives/memory cards) for longer term storage/recall.