bambulab/BambuStudio #6376: Online Models keeps appearing on the Home screen

HSMS-Rosmalen, 2025-04-01:
### Bambu Studio Version

2.00.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1S with AMS

### How to reproduce

Just go to the Home page (and have Online Models deselected in the Preferences). It just comes back every time you start the application.

### Actual results

On HOME page: Online Models are visible.

### Expected results

On HOME page: Online Models not visible.

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

realzorro, 2025-06-25:
Thank you for providing feedback on this issue. We are currently working on fixing it. Please stay tuned for updates in future versions. Thank you

MackBambu, 2025-07-28:
Similar issues can be addressed collectively here.