bambulab/BambuStudio #6377: Opening sliced plate 3mf immediately closes

davidjdixon, 2025-04-01:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macos 15.4

### Printer

H2D

### How to reproduce

1. Export any sliced model using 'Export plate sliced file'
2. Open exported plate file into Bambu Slicer
3. Note the message about 'After completing your operation File XYZ will be closed and create a new project' (which should actually say "and a new project will be created"). Click OK.

### Actual results

Project immediately closes and a new project starts in the 'prepare' tab.

### Expected results

Exported plate should load in the 'preview' stage, pre-sliced and ready to print.

### Project file & Debug log uploads

N/A - the same issue occurs with any pre-sliced file.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-04-02:
Hello, could you provide a screen recording to describe the issue? This would be helpful for us to reproduce and solve the problem.

bambu123, 2025-04-21:
> @BoBoYu-1<https://github.com/BoBoYu-1> screen recording attached
> screen-recording.mp4.zip<https://github.com/user-attachments/files/19568625/screen-recording.mp4.zip>

This should be normal behavior. Opening a slice project will close the old project.

davidjdixon, 2025-04-21:
> This should be normal behavior. Opening a slice project will close the old project.

@bambu123 This is incorrect and maybe I wasnt clear enough in the description and video recording, but the project closes "itself", not any pre-existing project.

bambu123, 2025-04-21:
Open a slice project exported from the current project, also as a new project.

davidjdixon, 2025-05-25:
> Open a slice project exported from the current project, also as a new project.

Sorry, what?