bambulab/BambuStudio #6379: Enter Key

ahansen9911, 2025-04-01:
### Bambu Studio Version

2.0.0.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1S

### How to reproduce

1. When scaling, moving, rotating, cutting.
2. The "Enter" key on the far right, is disabled while using Bambu Studio

### Actual results

I have to use the enter key by the alphabet.

### Expected results

It should enter.

### Project file & Debug log uploads

[debug_Mon_Mar_31_03_38_46_23684.log.zip](https://github.com/user-attachments/files/19551597/debug_Mon_Mar_31_03_38_46_23684.log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

guanyun-gudujian, 2025-04-02:
will be fixed next version

guanyun-gudujian, 2025-04-16:
@ahansen9911 
fixed
pls try the latest version: https://github.com/bambulab/BambuStudio/releases/download/v02.00.02.57/Bambu_Studio_win_public-v02.00.02.57-20250415200919.exe