bambulab/BambuStudio #6381: 更新到2.0.1版本的studio后，多设备管理废了

muyir123, 2025-04-01:
### Bambu Studio Version

2.0.1

### Where is the application from?

Bambu Lab github releases, Bambu Lab Official website

### OS version

WIN1123H2

### Printer

A1C

### How to reproduce

更新了最新的2.0.1studio后，切片后发送到多设备选项中启用AMS开关消失，完全无法使用AMS。
对于家庭多台机器，每台都配备了AMS，无外挂料盘，这个选项消失后相当于无法正常打印，现在只能退回到1.10.02.64使用。
但是之后2.0的studio切片的3mf文件，用1.0的只能获取几何数据，继续用老版本studio后续会非常麻烦，希望官方尽快修复

### Actual results

启用AMS开关消失

### Expected results

启用AMS开关正常出现，并且可以选择ams槽位，类似1.10.02.64及之前版本那样

### Project file & Debug log uploads

无需日志，可直接复现

### Checklist of files to include

- [x] Log file
- [x] Project file

muyir123, 2025-04-03:
@walterwongbbl @BoBoYu-1 这个bug会修复吗？

walterwongbbl, 2025-04-09:
@muyir123 
We will fix this issue

muyir123, 2025-04-09:
> 我们将解决此问题

thanks