bambulab/BambuStudio #6384: 2.0.0 and 2.0.1 give a never ending error dialog box.

ilikecheese, 2025-04-01:
### Bambu Studio Version

2.0.0 and 2.0.1

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 11

### Additional system information

13th gen core i9
Dell precision 5680
32 GB Ram
Nvdia RTX 2000 GPU

### Printer

X1 Carbon w/ AMS

### How to reproduce

Open program.

### Actual results

[image]

### Expected results

No error dialog.

### Project file & Debug log uploads

Updated Handy App. Deleted and reinstalled handy app.
Studio - no project file. fresh open. fresh install. complete uninstall, reinstall. 
All scenarios lead to the dialog.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ilikecheese, 2025-04-02:
Today, 2.0.0 has not given me an error dialog. I have no idea of what could be different. I did not change anything. 

I will try to install 2.0.1 and give an update.

lanewei120, 2025-04-02:
Hello @ilikecheese 

could you help to upload your log files here?

lanewei120, 2025-04-07:
not this one
you can find them from
[image]

ilikecheese, 2025-04-10:
[log.zip](https://github.com/user-attachments/files/19689354/log.zip)

Good luck!

lanewei120, 2025-04-19:
@StoneLiBambulab 
seems some network errors

Hello @ilikecheese 
is this issue still there

closed 2025-08-30 by ilikecheese