bambulab/BambuStudio #6385: Bambu Studio Print Preview Not Showing

ERIC-Y-11, 2025-04-01:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Home 24H2

### Additional system information

CPU: Ryzen 5 5500U
GPU: Radeon Integrated Graphics
RAM: 16GB DDR4

### Printer

A1

### How to reproduce

Open a 3MF file

### Actual results

Black screen where the preview should be

### Expected results

3D Print on the build plate as a preview

### Project file & Debug log uploads

No debug log

### Checklist of files to include

- [ ] Log file
- [ ] Project file

guanyun-gudujian, 2025-04-02:
@xeric6   disable amd image sharpening and try again
https://www.amd.com/en/products/software/adrenalin/radeon-image-sharpening.html

xeric6, 2025-04-02:
tysm

Mauriskess, 2025-06-28:
> > tysm
> 
> is it work now?

It fixed it for me, thanks!