bambulab/BambuStudio #6386: Numpad enter does not apply rotation to models

justlovett0, 2025-04-01:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Version 10.0.19045 Build 19045

### Printer

Bambu P1S

### How to reproduce

Load model into Prepare on any plate
Click Rotate or hotkey "R"
Type in rotational value on any access

Numpad ENTER does nothing. You must use 10keyless ENTER above right shift.

[image]

### Actual results

Object does not apply rotational coordiantes/values.

### Expected results

Object should apply entered rotational coordinates/values.

### Project file & Debug log uploads

[BugReport Project File ZIP.zip](https://github.com/user-attachments/files/19558286/BugReport.Project.File.ZIP.zip)

[Debug Files Latest.zip](https://github.com/user-attachments/files/19558295/Debug.Files.Latest.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

guanyun-gudujian, 2025-04-02:
will be fixed next version

guanyun-gudujian, 2025-04-16:
@justlovett0 
fixed
pls try the latest version: https://github.com/bambulab/BambuStudio/releases/download/v02.00.02.57/Bambu_Studio_win_public-v02.00.02.57-20250415200919.exe

justlovett0, 2025-04-16:
> [@justlovett0](https://github.com/justlovett0) fixed pls try the latest version: https://github.com/bambulab/BambuStudio/releases/download/v02.00.02.57/Bambu_Studio_win_public-v02.00.02.57-20250415200919.exe

Yes it is fixed. Thank you.

closed 2025-04-16 by justlovett0