bambulab/BambuStudio #6388: H2D 0.12mm layer height profile messed up

ffafe5, 2025-04-02:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Printer

H2D

### How to reproduce

The 0.12mm layer height profile is messed up for H2D. The walls are doubled in quantity.

### Actual results

See here for a comment made documenting this issue in a past issue thread that was closed prematurely: https://github.com/bambulab/BambuStudio/issues/6169#issuecomment-2767885373

### Expected results

0.12mm layer height profile should use wall quantity setting intended for a 0.4mm nozzle but is actually picking a wall count intended for 0.2mm nozzles.

### Project file & Debug log uploads

:(

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Master-Mace, 2025-04-19:
Can confirm. Speed settings are turned way down, seems only the layerhights have been adjusted.

XunZhangBambu, 2025-05-10:
@ffafe5 @james7360 @Master-Mace   For H2D, the fine and extra fine configurations are set for print quality, and you can compare them to the high quality configuration of the X series.

james7360, 2025-05-10:
@XunZhangBambu have you tried to compare them?

Master-Mace, 2025-05-11:
@XunZhangBambu yes,please do a one on one comparison between the two printers.

And why not just set the same amount of profiles for the H2D as for the X1C?

ffafe5, 2025-05-11:
These responses from bambu are very unprofessional