bambulab/BambuStudio #6391: Update???????

hans7321, 2025-04-02:
### Bambu Studio Version

2

### Where is the application from?

Bambu Lab Official website

### OS version

win11

### Printer

a1

### How to reproduce

[image]

### Actual results

I have been strugling for 3 hours thanks to your latest "update" more like a downgrade, I had to remove all my bambu files files from my pc and reinstall the software and re install the printer with  the wifi setup 
A NORMAL update should not introduce that much trouble and even to send this complaint, I have to use  git hub? and not the bambu site very customer unfriendly. 

[image]

### Expected results

fix IT

[image]

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-04-03:
[image]
Hello, we are very sorry for the inconvenience caused by the upgrade version. We suggest that you use the free installation version of the Windows zip package and try installing the exe version again after a period of time. So, what problem have you encountered? Could you provide the log folder in this path so that we can see what problem you have encountered.

Haidiye00, 2025-04-03:
Hello, if the problem is difficult to describe, you can also record a GIF.can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

Haidiye00, 2025-07-22:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~
If there are still questions, we will analyze them again.

closed 2025-07-22 by Haidiye00