bambulab/BambuStudio #6392: layer shift due to ventilation

Alama32, 2025-04-02:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: dual xeon 2643V4
Ram: 256 GB
GPU: Nvidia RTX 3060 Ti

### Printer

Bambu Lab A1

### How to reproduce

1. Create an object with cavities or bridges
2. Cut the object out
3. View the layers in fan speed mode
4. You can see that the layers with bridges or supports are affected by higher fan speeds.

### Actual results

See the result on the attached image

[image]

### Expected results

Print the outer layers with the same settings throughout the print. Because you can clearly see that this impacts the entire layer.

[image]

### Project file & Debug log uploads

[brimcurler.zip](https://github.com/user-attachments/files/19567726/brimcurler.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

QingZhangBambu, 2025-04-03:
i will ask concerned staff to check this.

ScorpioLingDong, 2025-04-08:
We have tested the official PETG filament and did not observe any significant defects, as shown in the image below.

[image]

Currently, we suspect that the shrinkage behavior of your filament may be more sensitive to temperature. Therefore, we recommend disabling the fan pre_start feature in this case， which may be helpful.

[image]

In future versions, the fan pre_start feature will only be enabled for the official PLA and PETG filaments to enhance cooling performance.

Thanks for your feedback.

IlyaDin, 2025-10-25:
**Hi!**
I am facing the same problem. When the "Pre start fan time" option is enabled, even if it is set to 1 second, the entire layer and subsequent layers are printed with excessive cooling.

[image]