bambulab/BambuStudio #6394: Settingspage not checking screen resolution

swe47vao, 2025-04-02:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows11

### Printer

Bambu Lab P1S 

### How to reproduce

I have 2 monitors , one with 2560 x1440 resolution, the other with 1920 x1080.
Every time I open the settingspage (Ctrl -P)  on the screen with lower resolution , the page doesn't scale properly. That get the window to be outside the range of the screen, especially the top of the window which make it impossible to close or move the settings window. And when that window is open I can't get back to the Bambu studio.
Only way I've found to resolve it is to stop the activity for BambuStudio in Taskmanager.

### Actual results

 Bambustudio closes down and I have to restart it, move it to the larger window and open settings again on that screen

### Expected results

The settings window should not be able to get outside the range of the screen it's open on.
Possible other solutions is that it gets scrollable or other way to close the settings window

### Project file & Debug log uploads

N / A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-04-03:
Hello, could you provide a screen recording to describe the issue? This would be helpful for us to reproduce and solve the problem.

Haidiye00, 2025-04-03:
Hello, thank you for your feedback. We will add a cancel button at the bottom of the dialog to ensure that the window can exit. Is this okay？

Haidiye00, 2025-08-13:
https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.85
Please use the latest version, thank you

closed 2025-08-13 by Haidiye00