bambulab/BambuStudio #6398: Corrupted Filament preset - cannot remove or use

brain-slug2102, 2025-04-02:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.4

### Additional system information

issue occurs on a 2019 iMac with an intel i9 9900k and AMD Radeon GPU, 32GB RAM
issue also occurs on an Apple M4 Pro chip with 25GB Unified Memory

### Printer

Bambu Lab A1

### How to reproduce

Issue has been replicated on 2 Macs running the same versions of macOS and Bambu Studio
One of them completely crashes the Mac during reproduction.

After a successful long print, I went to print more and realised the AMS Lite only shows 3 filaments not 4. Trying to add the 4th filament back in through the devices tab doesn't work as it no longer shows the Custom Filament profile. Looking in my custom filament window, the filement is there but there are no nozzle profiles. Clicking "add preset" under the title "Filament presets under filament" brings up a box with 2 drop down menus, clicking these menus one of my Macs doesn't load any options, clicking the menus on the other Mac completely crashes Bambu Studio.

Clicking delete preset doesn't delete it
I have gone into the configuration files and delete any files related to the custom filament but it still remains in Bambu Studio

### Actual results

Either Bambu Studio crashes or I cannot use, remove or delete the filament profile

[image]
[image]
[image]

### Expected results

The filament profile should either be working and able to be used for printing, or be delete and allow me to start again.

### Project file & Debug log uploads

Unable to upload project file, ive shared it on iCloud via the link below
https://www.icloud.com/iclouddrive/089hFcrBLXRPYMg8eT9rk1R8A#project
[log.zip](https://github.com/user-attachments/files/19574492/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

PeterKrul, 2025-05-26:
I have the same issue.

roadshow69, 2025-07-28:
I'm also having this issue.