bambulab/BambuStudio #6400: opening a rpoject in makerworld opens orca slicer not bambu studio

aaronbott, 2025-04-02:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

sequia 15.3.2

### Printer

bambu lab p1s

### How to reproduce

goto makerworld
pick ANY model
select open in bambu studio

### Actual results

when you do the above and have orca slicer and bambu studio installed on the same mac, the model will open orca slicer and not bambu studio

### Expected results

i wouldexpect the model to open in bambu studio

### Project file & Debug log uploads

this applies to all model on that have the option to "open in bambu studio"

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-04-03:
Hello, could you provide a screen recording to describe the issue? This would be helpful for us to reproduce and solve the problem.

aaronbott, 2025-04-03:
[image]

walterwongbbl, 2025-04-16:
Hi. @aaronbott
 This is because Orca uses our registry information. You can reinstall Bambu Studio, which will overwrite the registry information. This is due to the competitive relationship between our software and Orca.

closed 2025-04-16 by walterwongbbl

dmartzol, 2025-04-24:
> Hi. [@aaronbott](https://github.com/aaronbott) This is because Orca has tampered with our registry information. You can reinstall Bambu Studio, which will overwrite the registry information.

I don't think this is right. On Macos reinstalling is as simple as copying the Bambu Studio app into the Applications/ dir. I've done this multiple times during upgrades and it didn't fix the issue