bambulab/BambuStudio #6401: ENTER key on number pad no longer works on Windows 10 version of Bambu Studio 2.x

TheBigDavester, 2025-04-02:
### Bambu Studio Version

2.1.0.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Home 22H2

### Printer

A1

### How to reproduce

Change print scale value and hit num pad enter. Nothing. Worked in last version.

### Actual results

Change print scale value and hit num pad enter. Nothing. Worked in last version.

### Expected results

Change print scale value and hit num pad enter. Nothing. Worked in last version.

### Project file & Debug log uploads

Change print scale value and hit num pad enter. Nothing. Worked in last version.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-04-03:
Thank you for your feedback. We are addressing it and it will be included in the next version.

guanyun-gudujian, 2025-04-16:
@TheBigDavester 
fixed
pls try the latest version: https://github.com/bambulab/BambuStudio/releases/download/v02.00.02.57/Bambu_Studio_win_public-v02.00.02.57-20250415200919.exe

closed 2025-10-15 by Haidiye00