bambulab/BambuStudio #6402: after update

djwest9999, 2025-04-02:
### Bambu Studio Version

1.04.00.00

### Where is the application from?

Bambu Lab Official website

### OS version

Intel(R) Pentium(R) Silver N5030 CPU @ 1.10GHz   1.10 GHz

### Printer

A1

### How to reproduce

find stl
import..... wait
color ....... wait
slice........wait
wait
wait
then slowly go to print

### Actual results

so after the last update it is really slow to load any models, and just the feel is sluggish

### Expected results

work faster

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-04-03:
Hello, I feel that the version you are using is a bit outdated. Our latest version is already V2.0
Also, could you provide this STL model. Thank you
双语对照
笔记

Haidiye00, 2025-04-29:
Hello, please use the latest version, thank you.
https://github.com/bambulab/BambuStudio/releases/tag/V02.00.03.54

djwest9999, 2025-04-29:
So far so good after the last update. My prime Tower for some reason was
off of the build plate. All I did was shut down bamboo and then turn it
back on logging back into it and it fixed it. But as of right now it looks
like it's working well.

Thank you.

closed 2025-04-29 by Haidiye00