bambulab/BambuStudio #6403: SVG to surface crashes Bambu Studio

despisedatlanta, 2025-04-02:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS Sequoia 5.3.2 (24D81)

### Additional system information

Apple M1 Max Macbook Pro
RAM - 32GB
Built in Liquid Retina XDR Display
16-inch (3456 × 2234)
Apple XDR Display P3-1600nits
Promotion refresh rate
207GB available on 1TB SSD

### Printer

Bambu X1 Carbon.  

### How to reproduce

Insert curved object
Select object and the add modifier/SVG
Once file is selected it'll pop up on the object with the box to set adjustments to size/rotation etc.
If I click the checkbox "Use Surface" to have the SVG wrap the image onto the curved surface the program will crash.
Sometimes it's instantaneous, others it'll give me a couple seconds....long enough to think "oh cool, it'll work this time" and then poof.  Crash.

Video of what's happening attached.

https://github.com/user-attachments/assets/b563901a-f6de-4475-b161-071fe6c598bb

And note: I can do everything above in this file with the SVG as long as I don't click "Use Surface" and try to wrap it on the curve.  I can insert it flat and the program does not crash.

### Actual results

The program crashes.  I attached a video of the entire process above. The PDF attached is the Crash Report it sends to Apple.

[BambuCrashReport.pdf](https://github.com/user-attachments/files/19577370/BambuCrashReport.pdf)

### Expected results

What SHOULD happen?  The program should not crash.  LOL

### Project file & Debug log uploads

[CrashFiles.zip](https://github.com/user-attachments/files/19577269/CrashFiles.zip)

The project file, the last log file and the .svg file are included in the .zip file.

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-04-03:
Hello, we have identified this issue and have confirmed that it is a problem with the third-party library on the Mac. We suggest that you temporarily try this feature on the Windows version. We will continue to analyze this issue~We are very sorry

despisedatlanta, 2025-04-03:
That’d be a great idea if I had a PC but as I don’t I’m going to have to
leave this one to you guys. 😂

Thanks!

Shayne Huff

678-283-8376 cell

PRIMATE - Atlanta, GA
www.facebook.com/primategrind

Haidiye00, 2025-04-03:
https://zhuanlan.zhihu.com/p/26009691548
Okay, but if you have sufficient hardware resources on your Mac, you can also try installing a Windows virtual machine.

despisedatlanta, 2025-04-03:
Oh I know. I run Parallels on my work Mac for Mac-specific apps. But at
$100+ per year for Parallels plus the cost of a Windows license to run a
program that is touted to run natively on Windows OR Mac (which works great
for everything else) that’s not going to be an option.

Thanks for the suggestion though. I’ll just use a different program that
works until you guys can fix your program.

Shayne Huff

678-283-8376 cell

PRIMATE - Atlanta, GA
www.facebook.com/primategrind

Erik262, 2026-08-29:
Hello from the feature this problem still persists and keeps crashing every time.

closed 2026-09-09 by Haidiye00