bambulab/BambuStudio #6404: Bambu Lab Studio v2.0.1.50 freezon when try to open a earlier version .3mf file.

viulogic, 2025-04-03:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11, 10.0.26100

### Additional system information

12th Gen Intel(R) Core(TM) i7-12700F，2100 Mhz
32G, mother board Asustek, Prime B660-Plus D4

### Printer

A1

### How to reproduce

file, 
open project, 
the studio pop up a message box: this file contains a customized material,
click "yes"

### Actual results

studio freezon

### Expected results

open the file

### Project file & Debug log uploads

[2025_04_02_bugreport.zip](https://github.com/user-attachments/files/19579407/2025_04_02_bugreport.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-1, 2025-04-03:
Hello, could you provide a screen recording to describe the issue? This would be helpful for us to reproduce and solve the problem.

viulogic, 2025-04-03:
Here is the video record. Basically, no matter how I try to open this file, the studio will be stuck forever.
https://github.com/user-attachments/assets/3abdf0f8-2289-4ad9-be84-5b514899d1e0

Haidiye00, 2025-04-07:
[image]
Hello, our internal version has been fixed and will be resolved in the next version~

Haidiye00, 2025-04-07:
[image]
Hello, I have checked again and tested this version to be normal

viulogic, 2025-04-07:
Great! Thank youn very much and have a great day!

viulogic, 2025-04-07:
That's very interesting. I have tried on both my personal desktop and the office desktop. The studio will stuck and shows "not respnding" as shown in the video. I have checked the studio is verion 2.0.1.50. 
My office desktop system info is:
OS Name	Microsoft Windows 10 Enterprise
Version	10.0.19045 Build 19045
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Manufacturer	Dell Inc.
System Model	Precision 3650 Tower
System Type	x64-based PC
System SKU	0A58
Processor	11th Gen Intel(R) Core(TM) i7-11700K @ 3.60GHz, 3600 Mhz, 8 Core(s), 16 Logical Processor(s)
The graphic card is Nvidia Quadro RTX 4000

My home desktop is using Win11, and with Nvidia 4070 super. 
I will send you details later today.

Haidiye00, 2025-04-16:
Hello, please use the latest version, thank you.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.02.57

viulogic, 2025-04-16:
It works great! Thank you very much!