bambulab/BambuStudio #6405: Crash ogni volta che faccio lo slicer di un file.

phaeryus, 2025-04-03:
### Bambu Studio Version

20150

### Where is the application from?

Bambu Lab github releases

### OS version

windows

### Additional system information

Nome SO	Microsoft Windows 11 Pro
Versione	10.0.26120 build 26120
Descrizione altro SO 	Non disponibile
Produttore SO	Microsoft Corporation
Nome sistema	3DMARTS_PC
Produttore sistema	Dell Inc.
Modello sistema	Precision 3660
Tipo sistema	PC basato su x64
SKU sistema	0A9F
Processore	13th Gen Intel(R) Core(TM) i9-13900K, 3000 Mhz, 24 core, 32 processori logici
Versione/data BIOS	Dell Inc. 2.23.0, 04/02/2025
Versione SMBIOS	3.6
Versione controller integrato	1.17
Modalità BIOS	UEFI
Produttore scheda di base	Dell Inc.
Prodotto scheda di base	0PRR48
Versione scheda di base	A01
Ruolo piattaforma	Desktop
Stato avvio protetto	Attivato
Configurazione PCR7	Necessaria elevazione dei privilegi per la visualizzazione
Directory Windows	C:\WINDOWS
Directory System	C:\WINDOWS\system32
Periferica di avvio	\Device\HarddiskVolume1
Locale	Italia
Hardware Abstraction Layer	Versione = "10.0.26100.1"
Nome utente	3Dmarts_PC\3dmar
Fuso orario	ora legale Europa occidentale
Memoria fisica installata (RAM)	32,0 GB
Memoria fisica totale	31,7 GB
Memoria fisica disponibile	15,4 GB
Memoria virtuale totale	63,7 GB
Memoria virtuale disponibile	38,3 GB
Spazio file di paging	32,0 GB
File di paging	C:\pagefile.sys
Protezione DMA kernel	Attivato
Sicurezza basata sulla virtualizzazione	In esecuzione
Sicurezza basata sulla virtualizzazione: proprietà richieste	
Sicurezza basata sulla virtualizzazione: proprietà di sicurezza disponibili	Supporto virtualizzazione di base, Avvio protetto, Protezione DMA, Codice UEFI di sola lettura, SMM Security Mitigations 1.0, Controllo dell'esecuzione basato su modalità, Virtualizzazione APIC
Sicurezza basata sulla virtualizzazione: servizi configurati	Integrità del codice applicata da hypervisor
Sicurezza basata sulla virtualizzazione: servizi in esecuzione	Integrità del codice applicata da hypervisor, Conversione impaginazione applicata dall'hypervisor
Criterio di Controllo delle applicazioni per le aziende	Imposto
Criterio per la modalità utente di Controllo applicazioni per le aziende	Disattivato
Supporto per la crittografia automatica del dispositivo	Necessaria elevazione dei privilegi per la visualizzazione
Rilevato hypervisor. Le funzionalità necessarie per Hyper-V non verranno visualizzate.	

### Printer

x1 Carbon

### How to reproduce

Apro il programma  la visualizzazione del file in 3d è normale come sempre, ma al momento che effettuo lo slicer, dopo qualche secondo si blocca e dopo pochi secondi il programma si chiude.
Un'altra anomalia e da quando ho installato la versione nuova tutte le altre stampanti che erano registrate in orca slicer, si sono spostate in bambu studio, come se si fossero sovrascritte e in orca slicer sono state eliminati settaggi materiali e stampanti con settaggi.

### Actual results

Il programma si chiude.

### Expected results

Dovrebbe effettuare lo slicer e successivamente mi dovrebbe dare la possibilità di avviare una stampa.

### Project file & Debug log uploads

[Debug bambu studio.zip](https://github.com/user-attachments/files/19585311/Debug.bambu.studio.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

BoBoYu-1, 2025-04-07:
Thank you for your feedback. Could you please provide the 3mf file?

phaeryus, 2025-04-07:
Succede con qualsiasi file 3mf, anche con quelli che in passato andavo lisci senza problemi.

Il giorno 7 apr 2025, alle ore 04:50, BoBoYu-1 ***@***.***> ha scritto:

Thank you for your feedback. Could you please provide the 3mf file?

QingZhangBambu, 2025-05-23:
@phaeryus Apologies for replying late. Could you please try this method? It has worked for most people.
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing