bambulab/BambuStudio #6408: Move object UP out of platform

agania, 2025-04-03:
I need automated support under whole object, so I need move object up but it keep falling down,

work around is to add fake object, group it with my object, etc etc, but its just work around  so I need feature to move object up out of plate

that's all and that's all

inorishizuka, 2025-04-03:
Is that not a just raft lol

agania, 2025-04-03:
> Is that not a just raft lol

no, the raft doesn't move the object up so the tree support can reach it from below, the raft is just for better traction, I need a function that will allow me to move the object up without magic, I've seen a few posts on Reddit where people asked how to move an object up from the bed. Here is a magic Im talking about https://makerworld.com/en/models/1276115-jajko-voronoi-egg
using raft ruin this print so I had to figure out a solution