bambulab/BambuStudio #6409: Please give us back the old job starting dialog

butimo, 2025-04-03:
The new UI for starting a print is horrible. Sorry to say this. The old dialog show the colors and just up to 4 checkboxes made it so easy to see an understand. In the BambuStudio 2 dialog, I have to scan the whole dialog, always open "Advanced Options" to see and/or disable AMS, even have to scroll in the window.

The old UI was super clean, fast to understand and pretty perfect easy to use. Didn't get, why things was overworked to make it so complicated and hard to understand. Eye scan left, right, up, down, icons, text, .... 

If you have a couple of various bambu printers, it's a pain to use. 

Please bring the old, clean and super easy UI back. May as a power user view together with the multi-device toggle.

lgarron, 2025-04-16:
Looks like this has been greatly improved in version 02.00.02.57. 🤩

[image]

butimo, 2025-04-16:
Much better now. Still would be nice to get checkboxes or toggles back to have a color indicator and less reading, but that's much better than before.

I like how the team is listening to the users ;-)

closed 2025-04-16 by butimo