bambulab/BambuStudio #6414: Major issue with lag, or app freezing if there is no network connection

Brannan-repo, 2025-04-03:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

i7-11700k, 32g ram

### Printer

Bambu P1S

### How to reproduce

[Scenario 1] 
1. Disable your network adapter in Device Manager
2. Load Bambu Studio
3. Use the left mouse button to just rotate an empty plate

App Lockup steps(Scenario 2):
1. If on a local network, disconnect router incoming WAN to keep PC network connection but with no internet.
2. Load Bambu Studio
3. User the left mouse button to just rotate an empty place

*** These issues do no seem to occur if you have Bambu Studio loaded before disconnecting internet/network

### Actual results

[Scenario 1]
Bambu Studio studders and freezes every 1s or less as you are rotating the plate.

[Scenario 2]
Bambu studio will lock up completely and will not respond and you have to force shut down the app

### Expected results

I expect Bambu Studio to function correctly regardless of network connection. Allowing me to use the app, export the 3mf and do a print.

Something is clearly wrong with how Bambu Studio continues to try to connect to the central servers in the background, causing the app to be very laggy at best, and completely unusable at worst. Possibly non-threaded network connection attempt? Or some kind of blocking code related to the network connection.

This all happened when my ISP had an outage, so scenario 2 was the first thing I noticed since I was still connected to my router, but with no internet.
Scenario 1 occured after I disabled my network adapter for testing other things.

### Project file & Debug log uploads

No files/logs needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-04-07:
Hello, please try the latest version V2.0.1.50 on our official website. If you encounter any problems, please feel free to inform us in time. Thank you.
https://bambulab.cn/zh-cn/download/studio

Haidiye00, 2025-04-09:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

Brannan-repo, 2025-04-10:
No thanks on downloading some sketchy software just for a gif. I created one on my phone instead.
It seem 2.0.1.50 fixed the issue of lag if the network adapter is unplugged, but there is still an issue if you don't have internet (if you have a router separate from your modem you can unplug the ethernet from your modem to simulate this).
The app still freezes for maybe 20 seconds because it's failing to connect to the Bambu servers but the code is blocking, which causes the app to hang if there is no internet.

[image]

Haidiye00, 2025-04-10:
Hello, this GIF is quite vague. Can you record it again on the computer. It's better to have more related operations. At present, I have only seen the result card owner, but I haven't seen much of the previous operations, making it difficult to analyze. Thank you for your understanding

Brannan-repo, 2025-04-20:
The gif I posted shows the app freezing after the steps I mentioned. But here is the ScreenToGif recording showing the exact same thing. As of this posting I have updated to the latest 2.0.2.57 version.

[image]

Here are the steps to reproduce:
1) Unplug your MODEM from the internet but not your ROUTER.
   ** Just unplugging your pc from the router does not reproduce the issue. Your PC needs to think the internet is still connected.
2) Open the app.

Haidiye00, 2025-04-21:
Hello, if you log out of the software account and avoid connecting to the machine, can the software open normally, insert a cube, and slice successfully

XinZhangBBL, 2025-04-21:
@Brannan-repo  
When the software freezes, you can go to Task Manager, right-click on BambuStudio, select "Create dump file" (DMP), and then upload the log file and the DMP file. So that we can focus on your problem.

XinZhangBBL, 2025-05-19:
Halo, the problem solved or not? You can reopen the issue if not resolved.

closed 2025-05-19 by XinZhangBBL