bambulab/BambuStudio #6415: locking up

emneumann, 2025-04-03:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Edition	Windows 11 Pro Version	24H2 Installed on	‎12/‎30/‎2024 OS build	26100.3624 Serial number	PF4P0ARM Experience	Windows Feature Experience Pack 1000.26100.66.0

### Additional system information

Device name	NLTS_Beast
Processor	13th Gen Intel(R) Core(TM) i7-13700HX   2.10 GHz
Installed RAM	16.0 GB (15.7 GB usable)
Device ID	995F71A9-054B-431D-96C4-DBDBD05BEAB7
Product ID	00330-80000-00000-AA331
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

### Printer

BL P1S

### How to reproduce

In multi monitor environment (laptop w/ 3 external monitors) with BS on a secondary monitor, click down carrot, preferences.  

### Actual results

program locks and the only recourse it to kill it in task manager.

The preferences window actually opens up behind the main program window and whenever the preferences window is open, no action can be performed on the main window, not resizing, not moving, not even closing.

### Expected results

main window should be at the very least, movable to expose the background preferences window.  better still, the preferences window should always open in the foreground.

### Project file & Debug log uploads

none required/available

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-04-07:
Hello, the preference settings window is a modal window, which means that operations must be completed first before operating the main interface.

emneumann, 2025-04-07:
Yes, I understand that. And if that window opens behind the main window when invoked, it is inaccessible and therefore locks up the whole program.