bambulab/BambuStudio #6417: Filament synchronisation issue (2 AMS)

ChristianBrand22, 2025-04-03:
### Bambu Studio Version

V2.0.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

OS: Windows 11

### Additional system information

When I load a part from Maker World with for example two colours and cklick then on synchronize then the colours match not to the colours shown in the device in Bambu Studio and the last one (nr. 8) is alway empty. I must add then one. 
It looks like then the first AMS1 filament is the second one from AMS-2 and the 2, 3, 4, 5 sould be 1, 2, 3, 4 also on lower. 

Bambu Lab P1S: 01.07.00.00 (Newest)
AMS HUB: 00.00.00.43  (Newest)
AMS-1: 00.00.06.49  (Newest)
AMS-2: 00.00.06.49 (Newest)

[image]
[image]

After sync...
[image]

### Printer

Bambu Lab P1S

### How to reproduce

Update to Bambu Studio 2.0.1.50

### Actual results

When I cklick on each arrow on top of the filament in the device windows and synchronize again, then it works.

### Expected results

Correct sync. 

### Project file & Debug log uploads

[Log.zip](https://github.com/user-attachments/files/19591504/Log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

FlewByYou, 2025-04-06:
Trying selecting Overwriting when you sync the filament instead

FlewByYou, 2025-04-06:
Trying selecting Overwriting when you sync the filaments instead

Haidiye00, 2025-04-07:
[image]
Hello, we suggest you use the coverage mode, which will result in 8 consumables.
The last one (nr. 8) is always empty. I must add then one. "This is because your two consumables are exactly the same.

[image]
You can also manually add a consumable and match it~

Haidiye00, 2025-04-25:
Hello, please use the latest version
https://github.com/bambulab/BambuStudio/releases/tag/V02.00.03.54