bambulab/BambuStudio #6418: Engineering Plate Issue In 2.0.1.50

thisprintedthing, 2025-04-03:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

When trying to slice for the engineering plate, and PETG (any of its various flavors:  PETG, PETG HF, etc.) is showing for my filament, I get the error stating that the engineering plate is not suggested for use printing filament x(whatever flavor of PETG is showing at that position in the AMS), and if I still want to print the job, to set the bed temp to a number that is not zero.  First of all, PETG is great on the engineering plate.  I do it all the time.  In fact, PETG is the first filament listed at the front of the build plate.  Second, in my case at the moment, I'm using PETG HF, and the bed temp on the engineering plate for this filament is set to 70 already, not zero.

Again, this seems to happen regardless of what flavor of PETG I use, whether it's HF, plain old PETG, or otherwise.

And synching the filament in Studio with my AMS also seems to be buggy, particularly when I'm going from PLA to something that is not PLA, like PETG.

### Printer

X1-Carbon

### How to reproduce

1.  Set build plate in Studio to engineering plate.
2. Put PETG into AMS.  (I normally print in PLA, so it might be going from PLA to PETG,)
3. Synchronize filaments in Studio with AMS.
4. Error comes up at bottom right saying that PETG is not good on the engineering plate, and the slice button is grayed out unless I make a change to the build plate temp, which is already set to 70 degrees.  Changing it to something else, like 75, seems to nullify the error.

### Actual results

Explained in above steps.  I would show a screen shot, but the message seems to have disappeared on its own, turning the slice button green once again.  I tried to reproduce the issue, and I cannot at this moment, but I'm sure that it will happen again, as it has been happening frequently since version 2 was launched.  I have also switched to PLA, and synched it in Studio, while leaving the build plate set to engineering, and it still does not give me a warning about PLA and the engineering plate, even though that actually is a bad combination.  So it seems that the logic for displaying this message and conditionally graying out the slice button is broken.

### Expected results

It should only give this warning and gray out the slice button if PLA (or another filament that is incompatible with the engineering plate) is showing in the filament slot and the engineering plate is selected.

### Project file & Debug log uploads

The project file is irrelevant, as this will happen on any project/model.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

thisprintedthing, 2025-04-05:
I was able to get a screen shot of this behavior.  I am attaching it here.

[image]

XunZhangBambu, 2025-04-10:
@thisprintedthing  Hello ,Is it always happening after syncing AMS?

thisprintedthing, 2025-04-10:
It seems like it’s been happening when I switch between PLA and PETG, and vice versa. What’s worse is that it tends to slice PLA prints with PETG settings, even though it initially shows PLA in the slicer. But then once I realize that my print is ruined, often quite some time later, I’ll look again at Studio and see that it is once again showing PETG, even though it previously showed PLA, and I have to synch it again to slice it with PLA settings. I know that there seems to be some variances and inconsistencies with what I’m reporting on this. It seems to be random when it does it and when it doesn’t. But the synching of filaments between Studio and the AMS seems to be broken in ver 2. I didn’t have these problems before updating to ver 2. Sent from my iPhone