bambulab/BambuStudio #6420: Multi-Device view in previous versions showed progress and estimated time.

skeller2015, 2025-04-03:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1C

### How to reproduce

Multi-Device tab

### Actual results

Device Status only shows "Printing"

### Expected results

In previous versions, Device Status showed progress and estimated print time.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

rosskinard, 2025-04-06:
I posted in the Bambu Community forums about this around 11 days ago and multiple people have reported the same, asking for it to be returned.

[https://forum.bambulab.com/t/bambu-studio-2-0-0-95-multi-device-no-longer-shows-progress/156094/1](url)

Hopefully we can get this fixed.

Thanks
Ross

walterwongbbl, 2025-04-14:
Hi. @rosskinard @skeller2015 
This is a bug, and we will fix this issue. If you need to control multiple printers, we recommend using the Bambu Printing Farm software.

rosskinard, 2025-04-18:
Issue persists into the new 2.0.2.57 version. Progress bar is visible at the beginning, but once it switches to the text for "Preheating" it never changes back to the progress bar.

[image]