bambulab/BambuStudio #6423: Bambu Lab Studio Crashes When Imported/Added OBJ From MakerWorld

drmiller, 2025-04-03:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.3.2

### Printer

Bambu Lab X1C

### How to reproduce

1. Generate an obj file from image in MakerWorld
2. Create New project in Studio
3. Import/Add obj file
4. Bambu Lab Studio Crashes

### Actual results

 Bambu Lab Studio Crashes

### Expected results

Work with slicer with Studio. There is no project file because the app crashes.

### Project file & Debug log uploads

[Archive.zip](https://github.com/user-attachments/files/19595593/Archive.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

chenweize1998, 2025-04-04:
Same problem

BurzumBlacker, 2025-04-04:
Hi 
I create a Tinkercad project and also ceashes.... but I fixed this way:

- Create a default cube (red) then export as TEST.obj
- Tinkercad generates a TEST.zip. 
- Decompress the TEST.zip. Inside the folder you will see TEST.obj and OBJ.mtl
- Rename OBJ.mtl to TEST.mtl

Now Bambu Studio 2 will open the file.... I think the *.obj and the *.mtl names must match.

chenweize1998, 2025-04-04:
Just verified that 1.10.2 works fine for me using the same obj

Haidiye00, 2025-04-07:
Hello, we have fixed it and will release it in the next version.
Currently, you can save 3mf after importing in version 1.10 and then open it in version V2.0~
It is recommended to download the installation free zip file for version 1.10~
so sorry

closed 2025-04-16 by Haidiye00