bambulab/BambuStudio #6426: Gcode Slider Doesn't Move

OhTink89, 2025-04-04:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 26100.3476

### Printer

Bambu P1S

### How to reproduce

When I slice a model, I want to view the Gcode. The slider on the bottom of the Preview does not move. This worked before the recent update.

### Actual results

Bottom slider does not move.

### Expected results

Bottom slider moved and I could see the gcode in previous versions.

### Project file & Debug log uploads

[Mast Cradle.v1.zip](https://github.com/user-attachments/files/19602568/Mast.Cradle.v1.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

BoBoYu-1, 2025-04-07:
Thank you for your feedback. I have tested the slicer and found that the Gcode slider can slide normally to view the Gcode. Could you please try it again on your end? Or is there any special operation you can tell me about?

[image]

OhTink89, 2025-04-07:
I had to reinstall Bambu Studio. Then the slider worked again.

 �

 �

 �

20250407114625_rec_.gif (view on web) <https://github.com/user-attachments/assets/ed799591-a721-4b86-a3d3-47fd05019ccc> 

20250407114625_rec_.gif (view on web) <https://github.com/user-attachments/assets/ed799591-a721-4b86-a3d3-47fd05019ccc>

BoBoYu-1, 2025-04-09:
Has your problem been solved now? Can this order be closed now?

OhTink89, 2025-04-09:
It was resolved by reinstalling the software.

Paul

closed 2025-04-10 by BoBoYu-1