bambulab/BambuStudio #6427: blank window when trying to login/register bambu studio

SunHill86, 2025-04-04:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

Bambu Lab P1S

### How to reproduce

Clicking on login/register on bambu studi, the page is blank

### Actual results

[image]

### Expected results

put in username and password, but nothing happens

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-04-07:
Thank you for your feedback. Could you try another network instead?

vkukk, 2025-04-07:
When choosing language under Bambu Studio settings, this issue is resolved.

mitakadan, 2025-04-07:
> When choosing language under Bambu Studio settings, this issue is resolved.

Yes! I can confirm this! Thanks, @vkukk

realzorro, 2025-04-16:
Please choose language under Bambu Studio settings, this issue is resolved.