bambulab/BambuStudio #6434: 2.0 Sets Ender 3 bed temp for PLA to 35C

pr3sidentspence, 2025-04-05:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

5700X
64GB

### Printer

Creality Ender 3 V2

### How to reproduce

1. select ender 3
2. select generic pla
3. open model
4. slice model
5. export gcode

### Actual results

`M140 S35 ; set final bed temp`
and
`M190 S35 ; wait for bed temp to stabilize`

### Expected results

these should be 60

### Project file & Debug log uploads

[Small Thread - Lite Multipoint.zip](https://github.com/user-attachments/files/19613983/Small.Thread.-.Lite.Multipoint.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

abra5umente, 2025-04-07:
I can confirm I can reproduce this with PETG filament (my own profile).
Set in Filament Settings:

[image]

Resultant gcode - note how the bed temp is set to 35, despite 85 being set in filament settings:

[image]

I have not altered the startup gcode in any way, only settings I have changed is speed and print quality settings for my Mini+.

Machine start G-code:
---
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S170 ; set extruder temp for bed leveling
M140 S[bed_temperature_initial_layer_single] ; set bed temp
M109 R170 ; wait for bed leveling temp
M190 S[bed_temperature_initial_layer_single] ; wait for bed temp
M204 T1250 ; set travel acceleration
G28 ; home all without mesh bed level
G29 ; mesh bed leveling 
M204 T[machine_max_acceleration_travel] ; restore travel acceleration
M104 S[nozzle_temperature_initial_layer] ; set extruder temp
G92 E0
G1 Y-2 X179 F2400
G1 Z3 F720
M109 S[nozzle_temperature_initial_layer] ; wait for extruder temp

; intro line
G1 X170 F1000
G1 Z0.2 F720
G1 X110 E8 F900
G1 X40 E10 F700
G92 E0

M221 S95 ; set flow
---

Not really relevant here, but machine end G-code:
---
G1 E-1 F2100 ; retract
{if max_layer_z < 180}G1 Z{min(max_layer_z+2, 180)} F720 ; Move print head up{endif}
G1 X178 Y178 F4200 ; park print head
{if max_layer_z < 180}G1 Z{min(max_layer_z+30, 180)} F720 ; Move print head further up{endif}
G4 ; wait
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M107 ; turn off fan
M221 S100 ; reset flow
M900 K0 ; reset LA
M84 ; disable motors
---

My workaround for now is to manually edit the gcode files to set the temp to 85 before uploading to my printer. Not a great solution.

billgeek, 2025-04-15:
Same happens with any non-Bambu printer. Tested this using the Voron Trident 300mm and the Voron 2.4 300mm machines. Any filament, any build plate results in 35C.

XunZhangBambu, 2025-04-21:
@pr3sidentspence  It's because the selected bed plate is wrong. We have found the problem, will fix it soon