bambulab/BambuStudio #6436: 导入gcode文件后卡死或者软件崩溃

funison, 2025-04-05:
### Bambu Studio Version

v02.00.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

WIN10

### Printer

H2D

### How to reproduce

1.点击gcode文件
2.拖放到bambustudio
3.出现鼠标转圈或者软件崩溃退出现象

### Actual results

软件卡死一般出现在拖入p1s机型切片出来的gcode文件
软件崩溃一般出现在拖入3D校准网站生成的gcode文件，例如https://teachingtechyt.github.io/calibration.html#retraction
http://retractioncalibration.com/

### Expected results

可以正常打印gcode文件

### Project file & Debug log uploads

[Project1.zip](https://github.com/user-attachments/files/19614817/Project1.zip)
[log1.zip](https://github.com/user-attachments/files/19614815/log1.zip)
[Project2.zip](https://github.com/user-attachments/files/19614818/Project2.zip)
[log2.zip](https://github.com/user-attachments/files/19614816/log2.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

renochew, 2025-05-19:
Just came across this problem.

 For those who don't read Chinese, in v1, you can drag and drop custom gcode to Bambu Studio and have it send it to printer. If you do it in v2, it would cause the app crashing. This is a show stopper for me, back to v1 for the time being.

funison, 2025-05-19:
> Just came across this problem.
> 
> For those who don't read Chinese, in v1, you can drag and drop custom gcode to Bambu Studio and have it send it to printer. If you do it in v2, it would cause the app crashing. This is a show stopper for me, back to v1 for the time being.

yeah, but H2D has no chance to fall back :(