bambulab/BambuStudio #6440: P1S print progress gets stuck during filament change on layer 1

volltz, 2025-04-05:
### Bambu Studio Version

2.00.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

1. Open the specific project attached
2. Slice plate 2
3. Print
4. Wait for 1st color print to finish
5. During Filamanet change print gets stuck

### Actual results

Printer starts the filament change, but before cuting the filament it gets stuck in the "heating nozzle" step

[image]

### Expected results

When slicing with the old Bambu Lab Version (before 2.0) print continuess at filament change without issues.
Unfortunately not every model behaves like this. So far i usually only had it with filament changes in layer 1. If the first layer is a solid colour and there are only filament changes later in the print it is working fine

### Project file & Debug log uploads

[Catan_Base+Game,+Base+Game+5-6+Player+Expansion,+Seafarers,+Seafarers+5-6+Player+Expansion (1).zip](https://github.com/user-attachments/files/19616209/Catan_Base%2BGame.%2BBase%2BGame%2B5-6%2BPlayer%2BExpansion.%2BSeafarers.%2BSeafarers%2B5-6%2BPlayer%2BExpansion.1.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

volltz, 2025-04-05:
Did a little bit more analyzing and removed all (for me) yellow coloured pieces from the plate and tried printing the orange ones alone to avoid the color change on layer 1.
Result:
The print does not start at all!

So the issue is either related to the specific filament type (Bambu PLA Matte - orange, cant remember the exact name) or my AMS configuration (4x AMS)
I tried the same print with the pre 2.0 version and everything works fine

walterwongbbl, 2025-04-14:
Hi. @volltz 
Is there any error or pop-up message on the software.

volltz, 2025-05-01:
Currently i am using 2.0.3.54 and did not run into this issue anymore. I will observe for my next multi-coloured prints the filament change behaviour.
Maybe it also helped that i switched around the filaments in my AMS's?